h2zero/NimBLE-Arduino #946: Build fails when building with PlatformIO

WallNutss, 2025-04-25:
Hi,

I'm unable to build NimBLE in VSCode with PlatformIO on my ESP32 Device. 

Little context here, so previously I try to migrate my code that was using BLE library by Kolban, but trying to switch to NimBLE to reduce the flash size and the heap consumption. 

Before, I had successfully built the project by simply adding the NimBLE library to PlatformIO project. But because my input of the purpose is too big, I got a Stack Overflow on the stack
```
***ERROR*** A stack overflow in task nimble_host has been detected.
```

So, after that, I thinking of increasing the stack size of the nimble host using the menuconfig of the project. After I set the Bluetooth Nimble Host Stack size, and get it to be build. I get an error when building the project!

Here's my platform.ini
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino, espidf
monitor_speed = 115200
board_build.flash_size = 4MB
board_build.partitions = custom_partitions.csv
build_flags = 
	!python ./scripts/get_git_tag.py
lib_deps = 
	adafruit/Adafruit Fingerprint Sensor Library@^2.1.3
	bblanchon/ArduinoJson@^7.3.1
	adafruit/Adafruit PN532@^1.3.4
	tzapu/WiFiManager@^2.0.17
	h2zero/NimBLE-Arduino@^2.2.3
```

This is my device
Windows 11
PlatformIO: 6.1.18
VScode: 1.99.3
Device : ESP32

And this is the last build log
```prolog
Linking .pio\build\esp32dev\firmware.elf
c:/users/junin/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/esp32dev/esp-idf/bt/libbt.a(ble_hs_pvcy.c.o): in function `ble_hs_pvcy_remove_entry':
C:/Users/Junin/.platformio/packages/framework-espidf@3.40407.240606/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_pvcy.c:72: multiple definition of `ble_hs_pvcy_remove_entry'; .pio/build/esp32dev/lib001/libNimBLE-Arduino.a(ble_hs_pvcy.c.o):D:\PlatfromIO\Biometric-RFID-Access-Firmware/.pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_pvcy.c:87: first defined here
c:/users/junin/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/esp32dev/esp-idf/bt/libbt.a(ble_hs_pvcy.c.o): in function `ble_hs_pvcy_add_entry':
C:/Users/Junin/.platformio/packages/framework-espidf@3.40407.240606/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_pvcy.c:153: multiple definition of `ble_hs_pvcy_add_entry'; .pio/build/esp32dev/lib001/libNimBLE-Arduino.a(ble_hs_pvcy.c.o):D:\PlatfromIO\Biometric-RFID-Access-Firmware/.pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_pvcy.c:168: first defined here
c:/users/junin/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/esp32dev/esp-idf/bt/libbt.a(ble_hs_pvcy.c.o): in function `ble_hs_pvcy_ensure_started':
C:/Users/Junin/.platformio/packages/framework-espidf@3.40407.240606/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_pvcy.c:183: multiple definition of `ble_hs_pvcy_ensure_started'; .pio/build/esp32dev/lib001/libNimBLE-Arduino.a(ble_hs_pvcy.c.o):D:\PlatfromIO\Biometric-RFID-Access-Firmware/.pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_pvcy.c:198: first defined here
c:/users/junin/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/esp32dev/esp-idf/bt/libbt.a(ble_hs_pvcy.c.o): in function `ble_hs_pvcy_set_our_irk':
C:/Users/Junin/.platformio/packages/framework-espidf@3.40407.240606/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_pvcy.c:208: multiple definition of `ble_hs_pvcy_set_our_irk'; .pio/build/esp32dev/lib001/libNimBLE-Arduino.a(ble_hs_pvcy.c.o):D:\PlatfromIO\Biometric-RFID-Access-Firmware/.pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_pvcy.c:282: first defined here
c:/users/junin/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/esp32dev/esp-idf/bt/libbt.a(ble_hs_pvcy.c.o): in function `ble_hs_pvcy_our_irk':
C:/Users/Junin/.platformio/packages/framework-espidf@3.40407.240606/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_pvcy.c:274: multiple definition of `ble_hs_pvcy_our_irk'; .pio/build/esp32dev/lib001/libNimBLE-Arduino.a(ble_hs_pvcy.c.o):D:\PlatfromIO\Biometric-RFID-Access-Firmware/.pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_pvcy.c:345: first defined here
c:/users/junin/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/esp32dev/esp-idf/bt/libbt.a(ble_hs_pvcy.c.o): in function `ble_hs_pvcy_set_mode':
C:/Users/Junin/.platformio/packages/framework-espidf@3.40407.240606/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_pvcy.c:283: multiple definition of `ble_hs_pvcy_set_mode'; .pio/build/esp32dev/lib001/libNimBLE-Arduino.a(ble_hs_pvcy.c.o):D:\PlatfromIO\Biometric-RFID-Access-Firmware/.pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_pvcy.c:354: first defined here
c:/users/junin/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/esp32dev/esp-idf/bt/libbt.a(ble_hs_pvcy.c.o): in function `ble_hs_pvcy_enabled':
C:/Users/Junin/.platformio/packages/framework-espidf@3.40407.240606/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_pvcy.c:301: multiple definition of `ble_hs_pvcy_enabled'; .pio/build/esp32dev/lib001/libNimBLE-Arduino.a(ble_hs_pvcy.c.o):D:\PlatfromIO\Biometric-RFID-Access-Firmware/.pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_pvcy.c:372: first defined here
c:/users/junin/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/esp32dev/esp-idf/bt/libbt.a(ble_hs_pvcy.c.o): in function `ble_hs_pvcy_rpa_config':
C:/Users/Junin/.platformio/packages/framework-espidf@3.40407.240606/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_pvcy.c:308: multiple definition of `ble_hs_pvcy_rpa_config'; .pio/build/esp32dev/lib001/libNimBLE-Arduino.a(ble_hs_pvcy.c.o):D:\PlatfromIO\Biometric-RFID-Access-Firmware/.pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_pvcy.c:379: first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1
```

Any idea how to solve this or what I do wrong? I tried to see if there was something similar like this #745

djdehaan, 2025-04-25:
looking at the log: error is probably because you include libs from a header file, which itself is included serveral times in various other cpp or header files.
Add **#pragma once** on top of header file to avoid multiple inclusions, or move the lib inclusion to the cpp.

Jason2866, 2025-04-25:
Wrong setup in platformio.ini By `framework = arduino, espidf` you compile Arduino as an component of IDF. A working platformio.ini should look like this
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/54.03.20/platform-espressif32.zip
framework = arduino
monitor_speed = 115200
build_flags =
    '-DCONFIG_BT_NIMBLE_EXT_ADV=1'
lib_deps =
    https://github.com/h2zero/NimBLE-Arduino.git#nimble-1.6-c6-build-working
lib_ignore =
    BLE
    BluetoothSerial
    SimpleBLE
    WiFiProv
custom_component_remove =
    espressif/esp_hosted
    espressif/esp_wifi_remote
    espressif/esp-dsp
    espressif/network_provisioning
    espressif/esp_rainmaker
    espressif/rmaker_common
    espressif/esp_insights
    espressif/esp_diag_data_store
    espressif/esp_diagnostics
    espressif/libsodium
    espressif/esp-modbus
    espressif/esp-cbor
    espressif/esp-sr
    espressif/esp32-camera

[env:esp32s3]
board = esp32-s3-devkitc-1

[env:esp32c2]
board = esp32-c2-devkitm-1

[env:esp32c3]
board = esp32-c3-devkitm-1

[env:esp32c6]
board = esp32-c6-devkitm-1

[env:esp32h2]
board = esp32-h2-devkitm-1
```

h2zero, 2025-04-25:
@Jason2866 was almost right here. The problem is you have enabled nimble in menuconfig while using NimBLE-Arduino. This library was not designed for this because it includes the nimble stack, which the arduino core does not.

#### 3 options to fix this:
1. Keep nimble enabled in menuconfig and use [esp-nimble-cpp](https://github.com/h2zero/esp-nimble-cpp):
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino, espidf
monitor_speed = 115200
board_build.flash_size = 4MB
board_build.partitions = custom_partitions.csv
build_flags = 
	!python ./scripts/get_git_tag.py
lib_deps = 
	adafruit/Adafruit Fingerprint Sensor Library@^2.1.3
	bblanchon/ArduinoJson@^7.3.1
	adafruit/Adafruit PN532@^1.3.4
	tzapu/WiFiManager@^2.0.17
	h2zero/esp-nimble-cpp@^2.2.1          ; <-- same code but made for IDF use with nimble
```
2. Disable nimble in menuconfig and add the config changes to your platformio build flags:
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino, espidf
monitor_speed = 115200
board_build.flash_size = 4MB
board_build.partitions = custom_partitions.csv
build_flags = 
	!python ./scripts/get_git_tag.py
        -DCONFIG_BT_NIMBLE_HOST_TASK_STACK_SIZE=4096  ; <-- adjust this to whatever you need
lib_deps = 
	adafruit/Adafruit Fingerprint Sensor Library@^2.1.3
	bblanchon/ArduinoJson@^7.3.1
	adafruit/Adafruit PN532@^1.3.4
	tzapu/WiFiManager@^2.0.17
	h2zero/NimBLE-Arduino@^2.2.3
```
3. Change your code so to minimize what you do in your callbacks. This should probably be done regardless of the of how you proceed to fix this as it will hold up the BLE stack while your code runs in the callbacks.

WallNutss, 2025-04-27:
Hey @Jason2866 , thanks for help. There's a reason why I add espidf onto the framework settings. So little context here, previously my teams develop the firmware using only Arduino IDE, but it get harder to maintain those 1000+ line of code with no splitting the code whatsoever. But when I tried to port them using PlatformIO, I still need to port some of the library that are being used exclusive with using the Arduino (such as their AdafruitFingerprint). I tried to use the framework espidf only at the start, but I really had a hard time importing the Arduino and other Arduino Library as a component only, that's why I reference to this forum [here](https://community.platformio.org/t/arduino-as-espif-component-in-platformio-solution-that-works/36542)to use. Hence as you can see the current situation of my framework settings happen as I still want to access the ESP32 menuconfig. Do you have any guidance for this perhaps?

WallNutss, 2025-04-27:
Yes, thank you @h2zero for the reply. You're right — the library wasn't designed for this use case because it includes the NimBLE stack, which conflict when I turned on the NimBLE settings in the menuconfig. I try your first solution, and it works perfectly now! Thank you all for your help, really appreciate it :)

Jason2866, 2025-04-27:
And there is another option I have added in my fork `pioarduino` -> HybridCompile here you use as usual Arduino as framework and specify sdkconfig settings wanting to change from the choosen ones for standard Arduino. See example https://github.com/pioarduino/platform-espressif32/blob/dd2a7be34376513ad82353345115a5463475b5f4/examples/arduino-blink/platformio.ini#L65-L89

WallNutss, 2025-04-28:
> Well when using Arduino as an component of IDF why not using the https://github.com/h2zero/esp-nimble-cpp one? There is no unnecessary code which this lib has to work with standard Arduino. Since every Arduino part gets compiled, why not configure exactly as needed?

Yes, for now I leave the framework settings as it is, I only changed the library for NimBLE from NimBLE-Arduino to the esp-nimple-cpp like this 

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = espidf, arduino
monitor_speed = 115200
board_build.flash_size = 4MB
board_build.partitions = custom_partitions.csv
build_flags = 
	!python ./scripts/get_git_tag.py
lib_deps = 
	adafruit/Adafruit Fingerprint Sensor Library@^2.1.3
	bblanchon/ArduinoJson@^7.3.1
	adafruit/Adafruit PN532@^1.3.4
	tzapu/WiFiManager@^2.0.17
	h2zero/esp-nimble-cpp@^2.2.1
```

> See example https://github.com/pioarduino/platform-espressif32/blob/dd2a7be34376513ad82353345115a5463475b5f4/examples/arduino-blink/platformio.ini#L65-L89

Ahh, I see, I didn't know we can do it this way. Thank you for the reference!

closed 2025-04-28 by WallNutss