h2zero/NimBLE-Arduino #955: NimBLE HID + custom service

rryando, 2025-05-07:
Hi there, 

so I tried to create a service that act as I/O via customService from my phone to my esp32c3
the BLE also act as HID devices if connected (it could be connect silmutaneously) 

this works just fine when using bluedroid, but when ported into NimBLE, we got security issue to the HID not writing

- here's my *NimBLE* version of the script

```C++

void setup() {
    Serial.begin(115200);
    delay(1000);
    Serial.println("Starting ESP32 NimBLE Test...");

    NimBLEDevice::init(DEVICE_NAME);
    Serial.println("NimBLE Device initialized.");

    NimBLEDevice::setSecurityAuth(true, true, true);
    NimBLEDevice::setSecurityIOCap(BLE_HS_IO_DISPLAY_ONLY);
    NimBLEDevice::setSecurityPasskey(PASSKEY);
    NimBLEDevice::setMTU(500);

    pServer = NimBLEDevice::createServer();
    pServer->setCallbacks(&serverCallbacks);
    pHid = new NimBLEHIDDevice(pServer);

    pInputKeyboard = pHid->getInputReport(KEYBOARD_ID);   
    pOutputKeyboard = pHid->getOutputReport(KEYBOARD_ID); 
    pInputMediaKeys = pHid->getInputReport(MEDIA_KEYS_ID);

    pHid->setReportMap((uint8_t*)_hidReportDescriptor, sizeof(_hidReportDescriptor));
    pHid->setManufacturer("Manufacture"); // Set manufacturer
    pHid->setPnp(0x02, 0x1234, 0x5678, 0x0110); // Optional: PnP ID 
    pHid->setHidInfo(0x00, 0x01); // Set HID info with just version and country code

    pHid->startServices(); // This creates the HID service and characteristics

    // --- Setup Custom Service ---
    pCustomService = pServer->createService(CUSTOM_SERVICE_UUID);
    pCustomCharacteristic = pCustomService->createCharacteristic(
        CUSTOM_DATA_CHAR_UUID,
        NIMBLE_PROPERTY::READ |
        NIMBLE_PROPERTY::WRITE |
        NIMBLE_PROPERTY::WRITE_ENC |
        NIMBLE_PROPERTY::NOTIFY 
    );

    pCustomCharacteristic->setCallbacks(&charCallbacks); // Set callbacks for custom char
    pCustomCharacteristic->setValue("InitialNimValue");

  
    NimBLEDescriptor* pCCCDesc = pCustomCharacteristic->createDescriptor(
                                        "2902", // Standard CCCD UUID
                                        NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::WRITE_ENC,
                                        2 // Max length
                                    );
    pCustomCharacteristic->addDescriptor(pCCCDesc);
    pCustomService->start();
    Serial.println("Custom Service started.");

    // --- Advertising ---
    NimBLEAdvertising* pAdvertising = NimBLEDevice::getAdvertising();
    pAdvertising->addServiceUUID(CUSTOM_SERVICE_UUID);       // Add custom service UUID
    pAdvertising->addServiceUUID(pHid->getHidService()->getUUID()); // Add HID service UUID
    pAdvertising->setAppearance(0x03C1); // HID Keyboard appearance
    // pAdvertising->enableScanResponse(true);

    NimBLEDevice::startAdvertising();
    Serial.println("Advertising started (HID + Custom).");
    Serial.println("Setup Complete!");
}
```

-- and here's my *Bluedroid* version  --

```C++

void setup() {
    Serial.begin(115200);
    delay(1000); 
    Serial.println("Starting ESP32 BLE Test (Adjusted Conn Params)...");

    esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT);
   

    BLEDevice::init(DEVICE_NAME);
    Serial.println("BLE Device initialized.");

    pServer = BLEDevice::createServer();
    pServer->setCallbacks(&serverCallbacks); 

    pHid = new BLEHIDDevice(pServer);

    inputKeyboard = pHid->inputReport(KEYBOARD_ID);
    outputKeyboard = pHid->outputReport(KEYBOARD_ID);
    inputMediaKeys = pHid->inputReport(MEDIA_KEYS_ID);
    pHid->reportMap((uint8_t*)_hidReportDescriptor, sizeof(_hidReportDescriptor));
    pHid->manufacturer()->setValue("Manufacturer");

    // BLE Security configuration using BLESecurity class (as in previous step)
    BLESecurity *pSecurity = new BLESecurity();
    pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_MITM_BOND); 
    pSecurity->setCapability(ESP_IO_CAP_IO); 
    pSecurity->setKeySize(16);
    pSecurity->setInitEncryptionKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK);
    pSecurity->setRespEncryptionKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK);
    BLEDevice::setSecurityCallbacks(&securityCallbacks); 

    pHid->startServices();
    
    Serial.println("BLE Server created.");

    pCustomService = pServer->createService(CUSTOM_SERVICE_UUID);
    pCustomCharacteristic = pCustomService->createCharacteristic(
        CUSTOM_DATA_CHAR_UUID,
        BLECharacteristic::PROPERTY_READ |
        BLECharacteristic::PROPERTY_WRITE |
        BLECharacteristic::PROPERTY_NOTIFY 
    );

    BLE2902* p2902Descriptor = new BLE2902();
    p2902Descriptor->setAccessPermissions(ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE_ENCRYPTED);
    pCustomCharacteristic->addDescriptor(p2902Descriptor);
    pCustomCharacteristic->setCallbacks(&charCallbacks); 
    pCustomCharacteristic->setValue("InitialValue"); 

    pCustomService->start();

    BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
    if (pAdvertising) {
        pAdvertising->addServiceUUID(CUSTOM_SERVICE_UUID); 
        pAdvertising->addServiceUUID(pHid->hidService()->getUUID());
        pAdvertising->setAppearance(0x03C1); 
        Serial.println("Advertising configured (Custom + HID, adjusted conn params).");
        pAdvertising->setScanResponse(true);
        Serial.println("Advertising started.");
    } else {
        Serial.println("Error: Failed to get Advertising object!");
    }

    BLEDevice::startAdvertising();
    Serial.println("Setup Complete!");
}

```

things to note that, the `setAuthenticationMode` when advertising multiple service is tricky,
I assume the bluedroid one could complete bonding with `ESP_IO_CAP_IO` because of `onConfirmPIN` security callback did auto confirm,

but the `NimBLE` behaviour with this kinda difference, when I tried to connect, it immediately got disconnected before even pressing the pair prompt  (since there's no security callback to handle this (?))

i tried different `setSecurityIOCap` some could be connected both custom service + HID, but the HID was not triggering anything event after bonding with (`BLE_HS_IO_DISPLAY_YESNO`)

anyone have experienced this ?

thank you!

rryando, 2025-05-07:
Okay, so, I modified this library https://github.com/T-vK/ESP32-BLE-Keyboard to use the up-to-date version of the NimBLE
and test for simple sketch, it doesn't work to my android phone
but it does work on my windows pc 😅 

I'm even more confused now

rryando, 2025-05-08:
okay, so I have tried other HID lib (https://github.com/Mystfit/ESP32-BLE-CompositeHID) with minimal sketch + latest NimBLE
result: Keypress works, but Media Keys doesn't work in android (tested from NimBLE version 2.1.0 to latest)

I also tried the https://github.com/T-vK/ESP32-BLE-Keyboard with older version of Nimble (1.4.3) 
everything works in android

is there any workaround for the android media keys to works in latest NimBLE ?

rryando, 2025-05-08:
alright, it's confirmed if i downgrade to version **1.4.3** it works fine for both android and my windows pc (media keys) also works,
since this might be different context from what i report at the first place,i will close this issues, 

Thank you @h2zero ! Amazing Lib!

h2zero, 2025-05-08:
Hmm, downgrading versions isn't really a fix lol. I'm going to repoen this and ask that you test with the master branch.

rryando, 2025-05-08:
ah sure, at least the older version was working for me (better than bluedroid in terms of flash and ram 😅), 
so I planned to use it temporarily;

but yes, is the version in the master branch have potential fix already ? 
i will try the lattest commit on main and posted the result here later ya

rryando, 2025-05-08:
okay, i tested the code on the master, still not fixed

here's the log

the `Sending Volume Up via HID...` is the event where i sent the HID key to my paired android device, 
let me know if you need to to do more testing ya
```
          flags [READ|READ_ENC]
primary service
           uuid 0x180f
         handle 37
     end_handle 41
characteristic
           uuid 0x2a19
     def_handle 38
     val_handle 39
   min_key_size 0
          flags [READ|NOTIFY]
ccc descriptor
           uuid 0x2902
         handle 40
   min_key_size 0
          flags [READ|WRITE]
descriptor
           uuid 0x2904
         handle 41
   min_key_size 0
          flags [READ]
D NimBLEAdvertising: setAdvertisementData: 02 01 06 03 19 c1 03 03 03 4a 2a
D NimBLEAdvertising: << Advertising start
[   507][D][BleKeyboard.cpp:146] begin(): [] Advertising started!
D NimBLEServer: >> handleGapEvent: 
...
D NimBLEServerCallbacks: onPhyUpdate: default, txPhy: 2, rxPhy: 2
D NimBLEServer: >> handleGapEvent: 
...
D NimBLEServerCallbacks: onIdentity: default
...
D NimBLEServerCallbacks: onAuthenticationComplete: default
D NimBLEServer: << handleGapEvent
...
D NimBLEServerCallbacks: onConnParamsUpdate: default
...
I NimBLEServer: subscribe event; attr_handle=8, subscribed: true
...
D NimBLEServerCallbacks: onConnParamsUpdate: default
D NimBLEServer: << handleGapEvent
D NimBLEServer: Gatt Read event
...
I NimBLEServer: subscribe event; attr_handle=27, subscribed: true
D NimBLECharacteristicCallbacks: onSubscribe: default
...
I NimBLEServer: subscribe event; attr_handle=34, subscribed: true
D NimBLECharacteristicCallbacks: onSubscribe: default
D NimBLEServer: << handleGapEvent
D NimBLEDeviceCallbacks: onStoreStatus: default
D NimBLEServer: >> handleGapEvent: 
I NimBLEServer: subscribe event; attr_handle=39, subscribed: true
D NimBLECharacteristicCallbacks: onSubscribe: default
D NimBLEServer: << handleGapEvent
D NimBLEDeviceCallbacks: onStoreStatus: default
1
[ 14650][D][Tone.cpp:124] tone(): _pin=7, frequency=1800 Hz, duration=100 ms
[ 14657][V][Tone.cpp:62] tone_init(): Creating tone queue
[ 14662][V][Tone.cpp:68] tone_init(): Tone queue created
[ 14667][V][Tone.cpp:72] tone_init(): Creating tone task
[ 14673][V][Tone.cpp:85] tone_init(): Tone task created
Sending Volume Up via HID...
D NimBLEServer: Gatt Read event
D NimBLEServer: >> handleGapEvent:
D NimBLECharacteristicCallbacks: onStatus: default
D NimBLEServer: << handleGapEvent
[ 14680][D][Tone.cpp:31] tone_task(): Task received from queue TONE_START: _pin=7, frequency=1800 HzD NimBLEServer: Gatt Read event
D NimBLEServer: >> handleGapEvent:
D NimBLECharacteristicCallbacks: onStatus: default
D NimBLEServer: << handleGapEvent
, duration=100 ms
[ 14716][D][Tone.cpp:33] tone_task(): Setup LED controll on channel 0
E (15717) ledc: ledc_get_duty(745): LEDC is not initialized
[ 20676][D][Tone.cpp:124] tone(): _pin=7, frequency=1800 Hz, duration=100 ms
Sending Volume Up via HID...
D NimBLEServer: Gatt Read event
D NimBLEServer: >> handleGapEvent:
D NimBLECharacteristicCallbacks: onStatus: default
D NimBLEServer: << handleGapEvent
[ 20686][D][Tone.cpp:31] tone_task(): Task received from queue TONE_START: _pin=7, frequency=18D NimBLEServer: Gatt Read event
D NimBLEServer: >> handleGapEvent:
D NimBLECharacteristicCallbacks: onStatus: default
D NimBLEServer: << handleGapEvent
00 Hz, duration=100 ms
...