h2zero/NimBLE-Arduino #958: Bluetooth 5.0 - NimBLE_extended examples - Error compiling - ESP32C3.

ghmeteo, 2025-05-12:
Hello, trying to compile Bluetooth 5.0 NimBLE_extended code example, whether for Client, Server, etc., for the Xiao ESP32C3 module in ArduinoIDE and I get this error:

_ble_hs_periodic_sync.c: In function 'ble_hs_periodic_sync_free':
ble_hs_periodic_sync.c:58:24: error: 'struct ble_npl_event' has no member named 'event'
58 | if((psync->lost_ev).event != NULL)
| ^
exit status 1_

I checked the code in ble_hs_periodic_sync.c and the psync value in line 58 is not pointing to a _ble_npl_event_ type structure, but rather to a _ble_hs_periodic_sync_ type structure, so I don't understand the error message.

Any recommendations on what's going on?

Here the function code...
_void
ble_hs_periodic_sync_free(struct ble_hs_periodic_sync *psync)
{
    int rc;

    if (psync == NULL) {
        return;
    }

    if((psync->lost_ev).event != NULL)
        ble_npl_event_deinit(&psync->lost_ev);

#if MYNEWT_VAL(BLE_HS_DEBUG)
    memset(psync, 0xff, sizeof *psync);
#endif
    rc = os_memblock_put(&ble_hs_periodic_sync_pool, psync);
    BLE_HS_DBG_ASSERT_EVAL(rc == 0);
}_

ghmeteo, 2025-05-13:
Ok..
Is it supported for the Xiao nRF52840 board model?

h2zero, 2025-09-02:
Periodic advertising is not supported for any device at this point. This will be a version 3.x feature..

closed 2025-09-02 by h2zero