h2zero/NimBLE-Arduino #959: Mimicking multiple devices: bonding

vodka-bears, 2025-05-12:
Hello! I'm developing a gamepad that would work in 2 modes: xinput and with gyro. When booted in gyro mode the firmware increments the last byte of the mac address and uses othere metadata so they're different devices for both bluetooth stack and hid drivers on the computer. However problems arise when I bond the gamepad to the same computer in both modes. As soon as I bond in the other mode, the first gets unbonded as the bonding keys for the pc in nvram are overwritten. Are there any ways to work around this? Can I use several storages or extract the bonding data? I store it for only one client address, i.e. all the others are wiped when a new client is bonded.

h2zero, 2025-05-13:
What you are doing is operating 2 devices via one. This isn't really possible as bonding stores the identity address of the devices so that they are recognizable at all times and also known devices. By being 2 different devices with the same identity you are creating a situation that confuses the devices it's bonded to. It would depend on how forgiving those other devices are as to your success.

vodka-bears, 2025-05-13:
@h2zero thank you for your response. So you're basically saying that what I want is not supported by the library and no workarounds come to mind? I know there are other BLE devices namely gamepads that do exactly what I described. I presume they're not based on NimBLE-Arduino. Are you sure there's absolutely no way to make it work?

Perhaps something is possible to do with the nvs? Maybe there's a way to specify a different address in nvs for the library to store bonding data? Or maybe I can store the bonding data elsewhere and overwrite it if it doesn't match?

h2zero, 2025-05-15:
@vodka-bears There is some possibility of a way to do what you are seeking but it would require some in-depth handling of the bonding mechanics that are outside the scope of of the CPP API. You would need to override the lower level bond handling functions, which is available, but specialized and not something I would want to provide high level support for.

vodka-bears, 2025-05-21:
@h2zero I've been researching and thinking recently and it seems I've found a way to accomplish what I want. I'm thinking of making a PR and I want to clarify some things before doing the coding part so I don't work in vain.

What I see is that currently the NVS namespace name is hardcoded in ble_store_nvs.c. My idea is to make it customizable and to pass it as a defaultable parameter into NimBLEDevice::init(...). Is there any chance that would be accepted here or I need to submit to apache/mynewt-nimble first?

h2zero, 2025-05-22:
I don't think a PR for this would be appropriate here. What you can do is override the bonding functions and handle them for your needs though. The functions are global in scope, just change them after initialization.