h2zero/NimBLE-Arduino #961: Joy-Con is not listed / found during scan

LSatan, 2025-05-16:
Hi,
I want to connect my Nintendo Switch Joy-Con to esp32. I am using the NimBle_client example for this. The controller is not listed during the scan, but my smartphone finds it. It finds other devices in the area and I have already successfully connected an Xbox One controller. Why is the controller not found during scanning? Is there an additional function? I assumed that at least the Mac address would be displayed for all devices. Regardless of whether a connection is possible. Ide and libraries are up to date.
Do you have any idea what the problem could be?

Regards

h2zero, 2025-05-16:
If this device is showing in the Bluetooth menu on your phone but not in scans with nimble then it is a Bluetooth classic device, which is not supported by this stack.

LSatan, 2025-05-16:
Thank you for your quick reply. The request can then be closed.

closed 2025-05-19 by h2zero