h2zero/NimBLE-Arduino #962: 2.3.0 failing to compile

rhkean, 2025-05-20:
target:  esp32-Wroom
associated compiler flags:
*   -D CONFIG_NIMBLE_CPP_LOG_LEVEL=5
*   -D CONFIG_BT_NIMBLE_LOG_LEVEL=5
*   -D CONFIG_BT_ENABLED=1
*   -D CONFIG_BTDM_CTRL_MODE_BLE_ONLY=1
*   -D CONFIG_BT_BLUEDROID_ENABLED=0
*   -D CONFIG_BT_NIMBLE_ENABLED=1
*   -D CONFIG_BT_NIMBLE_ROLE_CENTRAL_DISABLED 
*   -D CONFIG_BT_NIMBLE_ROLE_OBSERVER_DISABLED
*   -D CONFIG_BT_NIMBLE_ROLE_BROADCASTER

going from 2.2.3 to 2.3.0 (no other changes) generated the following errors:

```
.pio/libdeps/MyProject/NimBLE-Arduino/src/NimBLEL2CAPChannel.cpp: In member function 'int NimBLEL2CAPChannel::writeFragment(std::vector<unsigned char>::const_iterator, std::vector<unsigned char>::const_iterator)':
.pio/libdeps/MyProject/NimBLE-Arduino/src/NimBLEL2CAPChannel.cpp:97:17: error: 'BLE_HS_EBADDATA' was not declared in this scope
         return -BLE_HS_EBADDATA;
                 ^~~~~~~~~~~~~~~
Retrieved `.pio/build/MyProject/lib739/NimBLE-Arduino/nimble/ext/tinycrypt/src/hmac_prng.c.o' from cache
.pio/libdeps/MyProject/NimBLE-Arduino/src/NimBLEL2CAPChannel.cpp:106:21: error: 'BLE_HS_ENOMEM' was not declared in this scope
             return -BLE_HS_ENOMEM;
                     ^~~~~~~~~~~~~
.pio/libdeps/MyProject/NimBLE-Arduino/src/NimBLEL2CAPChannel.cpp:106:21: note: suggested alternative: 'BLE_NPL_ENOMEM'
             return -BLE_HS_ENOMEM;
                     ^~~~~~~~~~~~~
                     BLE_NPL_ENOMEM
.pio/libdeps/MyProject/NimBLE-Arduino/src/NimBLEL2CAPChannel.cpp:120:18: error: 'BLE_HS_ESTALLED' was not declared in this scope
             case BLE_HS_ESTALLED:
                  ^~~~~~~~~~~~~~~
.pio/libdeps/MyProject/NimBLE-Arduino/src/NimBLEL2CAPChannel.cpp:127:18: error: 'BLE_HS_ENOMEM' was not declared in this scope
             case BLE_HS_ENOMEM:
                  ^~~~~~~~~~~~~
.pio/libdeps/MyProject/NimBLE-Arduino/src/NimBLEL2CAPChannel.cpp:127:18: note: suggested alternative: 'BLE_NPL_ENOMEM'
             case BLE_HS_ENOMEM:
                  ^~~~~~~~~~~~~
                  BLE_NPL_ENOMEM
.pio/libdeps/MyProject/NimBLE-Arduino/src/NimBLEL2CAPChannel.cpp:128:18: error: 'BLE_HS_EAGAIN' was not declared in this scope
             case BLE_HS_EAGAIN:
                  ^~~~~~~~~~~~~
.pio/libdeps/MyProject/NimBLE-Arduino/src/NimBLEL2CAPChannel.cpp:128:18: note: suggested alternative: 'EAI_AGAIN'
             case BLE_HS_EAGAIN:
                  ^~~~~~~~~~~~~
                  EAI_AGAIN
.pio/libdeps/MyProject/NimBLE-Arduino/src/NimBLEL2CAPChannel.cpp:129:18: error: 'BLE_HS_EBUSY' was not declared in this scope
             case BLE_HS_EBUSY:
                  ^~~~~~~~~~~~
.pio/libdeps/MyProject/NimBLE-Arduino/src/NimBLEL2CAPChannel.cpp:129:18: note: suggested alternative: 'BLE_NPL_EBUSY'
             case BLE_HS_EBUSY:
                  ^~~~~~~~~~~~
                  BLE_NPL_EBUSY
.pio/libdeps/MyProject/NimBLE-Arduino/src/NimBLEL2CAPChannel.cpp:141:13: error: 'BLE_HS_EREJECT' was not declared in this scope
     return -BLE_HS_EREJECT;
             ^~~~~~~~~~~~~~
Retrieved `.pio/build/MyProject/lib739/NimBLE-Arduino/nimble/ext/tinycrypt/src/sha256.c.o' from cache
.pio/libdeps/MyProject/NimBLE-Arduino/src/NimBLEL2CAPChannel.cpp:141:13: note: suggested alternative: 'BLE_NPL_ENOENT'
     return -BLE_HS_EREJECT;
             ^~~~~~~~~~~~~~
             BLE_NPL_ENOENT
```

h2zero, 2025-05-21:
Thank you @rhkean,I can confirm with those build flags that they will cause this issue, the workaround is this:
```
  ;-D CONFIG_BT_NIMBLE_ROLE_CENTRAL_DISABLED
  -D CONFIG_BT_NIMBLE_ROLE_OBSERVER_DISABLED
  -D CONFIG_BT_NIMBLE_ROLE_OBSERVER=0
```

I will look to fix this in the near term.

rhkean, 2025-05-21:
awesome!  thank you.

while I have your attention... :)
What are the correct flags for the following scenario
* esp32
* peripheral only (no client functionality needed)
* allow multiple bonds
* only single active connection
* as thin of a build as possible

h2zero, 2025-05-21:
I'm not sure why you have some of those flags defined, it looks like you are using espidf? 
The correct flags for what you want are simple to turn off obsever and central roles and if you will only have 1 connection, set the max connections to 1. 

I will be working on an update to address the build issues with these roles disabled.

Abasz, 2025-06-02:
I ma hitting the same compilation error with 2.3.0 with the below defines:

```
	'-D CONFIG_BT_NIMBLE_PINNED_TO_CORE=0'
	'-D CONFIG_BT_NIMBLE_ROLE_CENTRAL_DISABLED'
	'-D CONFIG_BT_NIMBLE_ROLE_OBSERVER_DISABLED'
```

I have these disabled so I can save some flash as my project do not use these functionalities.

h2zero, 2025-06-02:
Please see #964

Abasz, 2025-06-02:
Sorry, I missed this one. I confirm it is compiling with the fix branch

h2zero, 2025-06-03:
Closing this as resolved in #964

closed 2025-06-03 by h2zero