h2zero/NimBLE-Arduino #968: Handling duplicate scan result when filter mode is address and data

gkoh, 2025-05-25:
I've recently come across a device that actively advertises, then changes manufacturer data when set to pair.
After some poking about, I set the scan filter mode with:
```
NimBLEDevice::setScanFilterMode(CONFIG_BTDM_SCAN_DUPL_TYPE_DATA_DEVICE);
```
with the expectation that `onResult()` would be called due to a non-duplicate advertised device appearing.

This didn't happen, so debug logs got enabled.
It turns out the underlying esp framework does respect this scan mode, I can see the device being reported as 'new'.
However, after reviewing `NimBLEScan`, I think NimBLE-Arduino is only using the device address for its uniqueness check [here](https://github.com/h2zero/NimBLE-Arduino/blob/77f66f68c465ea43d9e8ab6ee701b30875d45c30/src/NimBLEScan.cpp#L92) and the log output:
```
Duplicate; updated: xxx
```
and thus `onResult()` does not get called when the data changes.

Is there a recommended way to handle this scenario?

For now I'm just starting the scan after the manufacturer data changes.

h2zero, 2025-05-25:
This doesn't sound right, probably a bug, but this situation is different that I have seen before. The data should be reported when complete and it should follow the filtering.

h2zero, 2025-05-27:
That makes sense, when the new advertisement comes it will not call onResult because it is not a scan response so you will need to catch it in the onDiscovered callback.

h2zero, 2025-06-02:
Interesting and good to know. In that case for this purpose it's probably best that you make a copy of the device when first detected and the use the onDiscovered callback and update the data in your copy if the scan response isn't changing. Alternatively stop the scan when it is seen a second time in onDiscovered, then restart it immediately to get a full report.

closed 2025-06-04 by gkoh