h2zero/NimBLE-Arduino #975: OnConnect() callback is not triggring

dinesh-slx, 2025-06-06:
I am using nimble version 2.2.3, with extended advertisiig i am trying to get the multiple connection for the very first time connect is going to one connect and advertising for the next device to connect but when i do second connect from different device it is not trigerring onConnect because of that i am unabe to do multiple connection and i have provice the serial logs also 
```
#include <NimBLEDevice.h>
#include <NimBLEExtAdvertising.h>
#include "ble.h"
#include "NimBLECharacteristic.h"
#include "NimBLEService.h"
#include "NimBLEUUID.h"
#include "bond.h"
#include "device_info.h"
#include "device_status.h"
#include "error.h"
#include "esp32-hal-log.h"
#include "esp32-hal.h"
#include "gatt.h"
#include "neopixel.h"
#include "ringtone.h"
#include <Adafruit_NeoPixel.h>
#include <string>
#include <strings.h>
// #include "device_info.h"

uint8_t primaryPhy = BLE_HCI_LE_PHY_1M;
uint8_t secondaryPhy = BLE_HCI_LE_PHY_1M;

NimBLEExtAdvertisement extAdv(primaryPhy, secondaryPhy);

/*

NimBLE-Arduino API doc:
https://h2zero.github.io/NimBLE-Arduino/

BLE Connection Parameters:
https://software-dl.ti.com/lprf/simplelink_cc2640r2_latest/docs/blestack/ble_user_guide/html/ble-stack-3.x/gap.html
https://www.btframework.com/connparams.htm
https://punchthrough.com/manage-ble-connection/
https://interrupt.memfault.com/blog/ble-throughput-primer

TODO: api for ble power control by central setting different connections parameters
- increase throughput for OTA firmware transfer

*/

static const char *TAG = "ble";
static uint16_t connHandle;

class ServerCallbacks : public NimBLEServerCallbacks {
  void onConnect(NimBLEServer *pServer, NimBLEConnInfo& connInfo) {
    NimBLEAddress centralAddr = connInfo.getAddress();
    ESP_LOGI(TAG, "central connected: %s", centralAddr.toString().c_str());
    connHandle = connInfo.getConnHandle();
    pServer->updateConnParams(connHandle,24,40,0,500);
    bool phyUpdated = pServer->updatePhy(connHandle, BLE_HCI_LE_PHY_1M_PREF_MASK, BLE_HCI_LE_PHY_1M_PREF_MASK, 0);
    ESP_LOGI(TAG, "PHY update %s", phyUpdated ? "success" : "failed");
    
    NimBLEDevice::startAdvertising(0);
  };

  void onDisconnect(NimBLEServer *pServer, NimBLEConnInfo& connInfo, int reason) {
    NimBLEAddress centralAddr = connInfo.getAddress();
    ESP_LOGI(TAG, "central disconnected: %s", centralAddr.toString().c_str());
    NimBLEDevice::startAdvertising(0);
  };
};

class CharacteristicCallbacks : public NimBLECharacteristicCallbacks {
  void onRead(NimBLECharacteristic *pCharacteristic, NimBLEConnInfo& connInfo) {
    
  };

  void onWrite(NimBLECharacteristic *pCharacteristic, NimBLEConnInfo& connInfo) {
  };
};

static CharacteristicCallbacks chrCallbacks;

void addServices(NimBLEServer *pServer);

NimBLEServer *InitBLE() {
  NimBLEDevice::init("VST-CL");

  // Get the device mac address
  NimBLEAddress address = NimBLEDevice::getAddress();

  // const uint8_t *native = address.getNative();
  const uint8_t *native = address.getBase()->val;

  // Compose unique device name by using the Lower Address Part (3 bytes) of bluetooth device
  // address, which uniquely identifies the device as long as we stick to ESP controllers.
  char buffer[14];
  sprintf(buffer, "VST-CL-%02X%02X%02X", native[2], native[1], native[0]);
  NimBLEDevice::setDeviceName(buffer);
  // NimBLEDevice::getAdvertising()->setName(buffer);
  ESP_LOGI(TAG, "Device name: %s", buffer);
  ESP_LOGI(
      TAG,
      "DeviceInfo: {Manufacturer:%s Serial: %s Model: %s HardwareVersion: %s FirmwareVersion: %s}",
      GetManufacturer(), GetSerial(), GetModel(), GetHardwareVersion(), GetFirmwareVersion());

  

  // Set BLE TX power
  // https://docs.espressif.com/projects/esp-idf/en/v4.4.6/esp32s3/api-reference/bluetooth/controller_vhci.html#_CPPv420esp_ble_tx_power_set20esp_ble_power_type_t17esp_power_level_t
  NimBLEDevice::setPower(ESP_PWR_LVL_P9); /** +9db*/
  NimBLEDevice::setSecurityAuth(true, false, true);
  NimBLEDevice::setSecurityIOCap(BLE_HS_IO_NO_INPUT_OUTPUT);

  NimBLEServer *pServer = NimBLEDevice::createServer();
  addServices(pServer);
  // NimBLEExtAdvertisement extAdv;

  extAdv.setName(buffer);
  extAdv.setConnectable(true);
  extAdv.addTxPower();
  extAdv.setFlags(0x06);
  // extAdv.addServiceUUID(NimBLEUUID(LIGHT_SERVICE));
  // extAdv.addServiceUUID(NimBLEUUID(AUDIO_SERVICE));
  // extAdv.addServiceUUID(NimBLEUUID(DEVICE_INFO_SERVICE));
  // extAdv.addServiceUUID(NimBLEUUID(DEVICE_STATUS_SERVICE));
  // extAdv.addServiceUUID(NimBLEUUID(BOND_SERVICE));

  extAdv.setMinInterval(0x20);
  extAdv.setMaxInterval(0x40);
  extAdv.setAppearance(0x0340); 
  return pServer;
}

void addServices(NimBLEServer *pServer) {
  pServer->setCallbacks(new ServerCallbacks());

  
  extAdv.setMinInterval(0x20);
  extAdv.setMaxInterval(0x40);
  extAdv.setAppearance(0x0340); 
}

bool StartAdvertising() {
  // NimBLEAdvertising *pAdvertising = NimBLEDevice::getAdvertising();
  // pAdvertising->addTxPower();
  // return pAdvertising->start();
  NimBLEExtAdvertising *pAdvertising = NimBLEDevice::getAdvertising();
  // Pass the configured advertisement to instance 0.
  pAdvertising->setInstanceData(0, extAdv);
  // Start advertising using instance ID 0.
  pAdvertising->start(0);
  return true;
}

uint16_t GetConnHandle() { return connHandle; }

void setup() {
  
  
  NimBLEServer *pServer = InitBLE();

  bool ok = StartAdvertising();

}

void loop() {delay(100);}

```

### Serial logs
```
[12:04:08:432] [ 10547][I][ble.cpp:48] onConnect(): [ble] central connected: e8:48:b8:c8:20:00␍␊
[12:04:08:441] [ 10558][I][ble.cpp:52] onConnect(): [ble] PHY update success␍␊
[12:04:08:450] D NimBLEServer: << handleGapEvent␍␊
[12:04:08:450] D NimBLEServer: >> handleGapEvent: ␍␊
[12:04:08:450] D NimBLEExtAdvertisingCallbacks: onStopped: Default␍␊
[12:04:08:454] D NimBLEServer: >> handleGapEvent: ␍␊
[12:04:08:459] D NimBLEServerCallbacks: onPhyUpdate: default, txPhy: 1, rxPhy: 1␍␊
[12:04:08:547] D NimBLEServer: >> handleGapEvent: ␍␊
[12:04:08:547] D NimBLEServer: << handleGapEvent␍␊
[12:04:08:553] D NimBLEServer: >> handleGapEvent: ␍␊
[12:04:08:553] I NimBLEServer: mtu update event; conn_handle=1 mtu=517␍␊
[12:04:08:563] D NimBLEServerCallbacks: onMTUChange(): Default␍␊
[12:04:08:563] D NimBLEServer: << handleGapEvent␍␊
[12:04:09:762] D NimBLEServer: >> handleGapEvent: ␍␊
[12:04:09:762] I NimBLEServer: subscribe event; attr_handle=8, subscribed: true␍␊
[12:04:09:771] D NimBLEServer: << handleGapEvent␍␊
[12:04:17:032] D NimBLEServer: >> handleGapEvent: ␍␊
[12:04:17:032] D NimBLEExtAdvertisingCallbacks: onStopped: Default␍␊
```

h2zero, 2025-06-11:
Please try with v2.3.0 as it may be resolved already.

Abasz, 2025-06-16:
As an additional update:

Also the NimBleServer->getConnectedCount() does not get updated. Almost like the device is not connected. But centrals can subscribe to notifications and receive data.

I tried this with an older garmin watch which does not support secure connection and for that onConnection is not triggered either but also no onAuthenticationComplete.

Here you can find two log files: (i) is for a forerunner 255 (has secure BLE) (ii) forerunner 645 (has no secure BLE)

Both gets a second connection from a PC at some point.

- [FR645.txt](https://github.com/user-attachments/files/20762469/FR645.txt)

I think onConnect is not getting called becuase connection seemingly fails:
D NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_CONNECT 
E NimBLEServer: Connection failed

But regardless the device completes subscription and receives notifications.

When you get to this line:
00:00:57.640948 VERBOSE - Device connected: 1

Actually device count should already be 2 as that is the second device.

- [FR255.txt](https://github.com/user-attachments/files/20762465/FR255.txt)

There is the same connection failure for the watch. But there is also onAuthenticationCompleted call. 

_on an probably unrelated note:_

For some reason I am not seeing debug logs on the ESP32S3 (ARDUINO_USB_CDC_ON_BOOT) even though I have enabled them. I was only able to get the logs on a firebeetle2 board (ESP32 wroom) 

```cpp
/** @brief Un-comment to set the debug log messages level from the NimBLE host stack.\n
 *  Values: 0 = DEBUG, 1 = INFO, 2 = WARNING, 3 = ERROR, 4 = CRITICAL, 5+ = NONE\n
 *  Uses approx. 32kB of flash memory.
 */
#define CONFIG_BT_NIMBLE_LOG_LEVEL  0

/** @brief Un-comment to set the debug log messages level from the NimBLE CPP Wrapper.\n
 *  Values: 0 = NONE, 1 = ERROR, 2 = WARNING, 3 = INFO, 4+ = DEBUG\n
 *  Uses approx. 32kB of flash memory.
 */
#define CONFIG_NIMBLE_CPP_LOG_LEVEL 5

/** @brief Un-comment to enable the debug asserts in NimBLE CPP wrapper.*/
// #define CONFIG_NIMBLE_CPP_DEBUG_ASSERT_ENABLED 1

/** @brief Un-comment to see NimBLE host return codes as text debug log messages.
 *  Uses approx. 7kB of flash memory.
 */
#define CONFIG_NIMBLE_CPP_ENABLE_RETURN_CODE_TEXT

/** @brief Un-comment to see GAP event codes as text in debug log messages.
 *  Uses approx. 1kB of flash memory.
 */
#define CONFIG_NIMBLE_CPP_ENABLE_GAP_EVENT_CODE_TEXT

/** @brief Un-comment to see advertisement types as text while scanning in debug log messages.
 *  Uses approx. 250 bytes of flash memory.
 */
#define CONFIG_NIMBLE_CPP_ENABLE_ADVERTISEMENT_TYPE_TEXT
```

h2zero, 2025-06-17:
Thank you for the details, that will help me to reproduce the issue, though I think this may be caused by something upstream. I'll look into this soon.

Abasz, 2025-06-21:
If you were to create a branch and update esp-nimble code to the latest master I am happy to test. 

Only thing is I need to understand what to change for espS3 CDC so debug logs show up. For some reason for S3 no logs are showing on the serial (serial is working as putting  Serial.printf specifically in works), while on a older Wroom (DRRobot's firebeetle 2) the same code shows serial debug from nimble

h2zero, 2025-06-21:
@Abasz This is very strange for sure, does this happen with version 2.2.3 or just 2.3.x?

h2zero, 2025-06-21:
@Abasz what happens if you disable all the security features?:
`NimBLEDevice::setSecurityAuth(false, false, false);`

h2zero, 2025-06-22:
Thanks @Abasz, that confirms the suspicion that this is an upstream bug, that was the major change in 2.3, so I will look into this and for now just used 2.2.3 until this is resolved.

h2zero, 2025-06-23:
@Abasz I decided to workaround this issue, it will apply to esp-idf users as well so it made sense to do so. Please test #987 and let me know if it's resolved.

Abasz, 2025-06-23:
> In the case for a server the status code should always be zero, this is an upstream bug where espressif changed the original code to always request DLE and BLE version, which are not always supported, so this workaround is valid and the app isn't missing anything.

Clear thanks! 

> This was already added here:

Actually I meant specifically log when you swallow the unsupported feature response, but I now understand based on your answer that this is probably not necessary.

h2zero, 2025-06-27:
All you need to do is start advertising after a connection is made because the connection process stops the advertising. You can do this for up to 3 connections by default or change the config for up to 9 connections.

dinesh-slx, 2025-06-27:
When i ran the example code,i am unable to see the device in bluetoothctl, do i need to add any extra flags? For extended advertising.

PR #987 (2025-06-28): [Bugfix](workaround) OnConnect not being called.
Upstream changes have resulted in a possible status of BLE_ERR_UNSUPP_REM_FEATURE, this resulted in the onConnect callback not being called despite the connection actually being created. This works around that bug to ensure that the connections are correctly tracked.

Fixes #975
closed 2025-06-28 by h2zero