h2zero/NimBLE-Arduino #977: can't connect after scan

icodk, 2025-06-10:
peripheral device that  also scan for other devices. However as soon as the device had started scanning it is no longer connectable.
It actually get connected but  the connection is terminated by the central(phone)
More information:
The sequence of events:
1. Device powerup and start advertising
2.  A Phone try to connect  and a pairing dialog is presented to the user if no bonding exist already (15:38:01.460)
3. User accept the connection and a secure connection is established (15:38:04.634)
4.  The device read CTS data from the phone periodically as long as the connection is on (15:38:05.131 - 15:38:05.680 and again in 15:38:17.138)
5. The user turns off the Bluetooth on the phone to disconnect from the device (15:38:27.892) and advertising is started again
6. The device start start scanning ( 15:38:27.892)
7. Scanning ended at 15:38:32.883
8.  User click on the bonding created above to reconnect 15:38:40.301
9.  the connection is aborted by the phone at                    15:38:40.600

To reconnect to the device the device is needed to be rest.
If I never start a scanning it works perfectly.
Also In the onConnect  call back I try to stop the scanning if it is on
I also check Encryption, Authentication and Bonding just before I call
                      `NimBLEDevice::startSecurity(connInfo.getConnHandle()); // force a security on the connection`
by calling `pConnInfo->isEncrypted()` and `pConnInfo->isAuthenticated()`  and `pConnInfo->isBonded()`
So I can understand that  Encryption and  Authentication are false but I would expect Bonded to be true

here is the printout
```
15:37:59.135 --> .
15:38:00.116 --> .
15:38:01.128 --> .
15:38:01.460 --> 
15:38:01.460 --> OnConnect: Client address: 84:ab:1a:e5:78:a3
15:38:01.460 --> 
15:38:01.460 --> ******* NOT ENCRYPTED  Connectin...
15:38:01.460 --> ******* NOT AUTHENTICATED  Connectin...
15:38:01.460 --> ******* NOT A BONDED Connectin...
15:38:01.643 --> 
15:38:01.643 --> MTU updated: 255 for connection ID: 0
15:38:04.634 --> 
15:38:04.634 -->  1 -> 2
15:38:04.634 --> Secured connection to: 84:ab:1a:e5:78:a3
15:38:05.131 --> 
15:38:05.131 --> Start CTS Read from  84:ab:1a:e5:78:a3
15:38:05.332 --> 
15:38:05.332 --> Service 1805 found...
15:38:05.580 --> 
15:38:05.580 --> Read permissions for Char 2A2B...
15:38:05.680 --> Total sec: 1749569881
15:38:05.680 --> CTS Read succeeded...
15:38:05.680 --> 
15:38:05.680 --> display_time()...
15:38:17.138 --> 
15:38:17.138 --> Start CTS Read from  84:ab:1a:e5:78:a3
15:38:17.320 --> 
15:38:17.320 --> Service 1805 found...
15:38:17.554 --> 
15:38:17.554 --> Read permissions for Char 2A2B...
15:38:17.686 --> Total sec: 1749569893
15:38:17.686 --> CTS Read succeeded...
15:38:27.892 --> 
15:38:27.892 -->  2 -> 3
15:38:27.892 --> Client disconnected - start advertising
15:38:27.892 --> 
15:38:27.892 --> ==== Start Scanning
15:38:29.123 --> .
15:38:30.138 --> .
15:38:31.119 --> .
15:38:32.132 --> .
15:38:32.883 --> 
15:38:32.883 --> Scan Ended; reason = 0
15:38:33.146 --> .
15:38:34.129 --> .
15:38:35.143 --> .
15:38:36.140 --> .
15:38:37.137 --> .
15:38:38.127 --> .
15:38:39.125 --> .
15:38:40.119 --> .
15:38:40.301 --> 
15:38:40.301 --> OnConnect: Client address: 84:ab:1a:e5:78:a3
15:38:40.301 --> 
15:38:40.301 --> ******* NOT ENCRYPTED  Connectin...
15:38:40.301 --> ******* NOT AUTHENTICATED  Connectin...
15:38:40.301 --> ******* NOT A BONDED Connectin...
15:38:40.600 --> 
15:38:40.600 --> Client disconnected - start advertising
15:38:41.115 --> .
```

h2zero, 2025-06-11:
It looks like the bond info is not being stored by the esp32 in your logs as it should have the bonded flag set on the next connection. Have you changed any config options? Please try again after full erasing the flash.

icodk, 2025-06-11:
I erased the chip
```
Configuring flash size...  
Erasing flash (this may take a while)...  
Chip erase completed successfully in 3.5 seconds. 
```
But it didn't help 
```

8:12:35.676 --> ==== Start Scanning
18:12:35.904 --> .
18:12:36.913 --> .
18:12:37.909 --> .
18:12:38.151 --> 
18:12:38.151 --> OnConnect: Client address: 84:ab:1a:e5:78:a3
18:12:38.151 --> 
18:12:38.151 --> >>>> Scan stop !
18:12:38.151 --> ******* NOT ENCRYPTED  Connectin...
18:12:38.151 --> ******* NOT AUTHENTICATED  Connectin...
18:12:38.151 --> ******* NOT A BONDED Connectin...
18:12:38.447 --> 
18:12:38.447 --> Client disconnected - start advertising
18:12:38.447 --> 
18:12:38.447 --> Disconnected Reason: 531
18:12:38.666 --> 
18:12:38.666 --> ==== Start Scanning
18:12:38.898 --> .
18:12:39.895 --> .
18:12:40.908 --> .
18:12:41.895 --> .
18:12:42.906 --> .
18:12:43.647 --> 
18:12:43.647 --> Scan Ended; reason = 0

```
I haven't change anything in the configuration.

icodk, 2025-06-12:
I tested it on another board (ESP32-C3 Super Mini) and it works great.
It is probably the specific board (which is of type:ESP32-WROOM-DA) as you suggested.
SO DON'T FIX IT :-)