h2zero/NimBLE-Arduino #980: compilation error "operator '&&' has no right operand" and "#if with no expression"

pabloandresm, 2025-06-11:
Since version 2.3.1 I am receiving both errors described in the title multiple times.
In my project I have "**CONFIG_BT_NIMBLE_ROLE_PERIPHERAL=y**" and "**CONFIG_BT_NIMBLE_ROLE_BROADCASTER=n**"

The **nimconfig.h** does this:
**/* Cannot use server without advertise */
#if defined(CONFIG_BT_NIMBLE_ROLE_PERIPHERAL) && !defined(CONFIG_BT_NIMBLE_ROLE_BROADCASTER)
#define CONFIG_BT_NIMBLE_ROLE_BROADCASTER
#endif**

And in several parts of the code (like NimBLEAdvertisementData.cpp) you have things like this:
**#if (CONFIG_BT_ENABLED && CONFIG_BT_NIMBLE_ROLE_BROADCASTER && !CONFIG_BT_NIMBLE_EXT_ADV) || defined(_DOXYGEN_)
...
#endif**

...and the error appears in the second && complaining about **CONFIG_BT_NIMBLE_ROLE_BROADCASTER** with **"operator '&&' has no right operand"**

There are 2 possible solutions here:
1) or either you do "**#define CONFIG_BT_NIMBLE_ROLE_BROADCASTER 1**" instead of "**#define CONFIG_BT_NIMBLE_ROLE_BROADCASTER**" in the nimconfig.h
2) or either you replace all appearances of "**#if (CONFIG_BT_ENABLED && CONFIG_BT_NIMBLE_ROLE_BROADCASTER && !CONFIG_BT_NIMBLE_EXT_ADV) || defined(_DOXYGEN_)**" by using "**defined(CONFIG_BT_NIMBLE_ROLE_BROADCASTER )**" instead of plain "**CONFIG_BT_NIMBLE_ROLE_BROADCASTER**"

h2zero, 2025-06-11:
Okay, I see now, this issue belongs in esp-nimble-cpp and I will fix it there now. 
The problem is that these were supposed to be removed but got missed:
```
/* Cannot use client without scan */
#if defined(CONFIG_BT_NIMBLE_ROLE_CENTRAL) && !defined(CONFIG_BT_NIMBLE_ROLE_OBSERVER)
#define CONFIG_BT_NIMBLE_ROLE_OBSERVER
#endif

/* Cannot use server without advertise */
#if defined(CONFIG_BT_NIMBLE_ROLE_PERIPHERAL) && !defined(CONFIG_BT_NIMBLE_ROLE_BROADCASTER)
#define CONFIG_BT_NIMBLE_ROLE_BROADCASTER
#endif
```

h2zero, 2025-06-12:
The reason to allow this is because you could be a central and a peripheral with no need to advertise. In your case you need to be able to advertise so you must enable the broadcaster role.

h2zero, 2025-09-02:
Closing this as resolved.

closed 2025-09-02 by h2zero