h2zero/NimBLE-Arduino #990: BT Home v2 format - could an example be added>

deeja, 2025-06-27:
[image]

BT Home uses the advertisements to broadcast data from very low power devices. 

It would help the Home Assistant / ESPHome community (and myself) if you were able to help with an example for https://bthome.io/format/

I'm working it out at the moment, but a gold standard example would be amazing.

h2zero, 2025-06-28:
Not much of an example necessary for this:
```
auto pAdv = NimBLEDevice::getAdvertising();
pAdv->setServiceData((uint16_t)0xFCD2, <your data array here>);
```

deeja, 2025-06-28:
I found something that seems to be appropriate here https://github.com/Chreece/BTHomeV2-ESP32-example

closed 2025-06-28 by deeja

h2zero, 2025-07-02:
Thanks, I'll pin this for anyone else that needs to see it.