h2zero/NimBLE-Arduino #1001: Redefinition warnings in esp32 core 3.3.0

ednieuw, 2025-07-25:
I got these warnings with core 3.3.0

```
c:\Users\xxxxx\Documents\Files\Arduino\libraries\NimBLE-Arduino\src/nimconfig.h:254:9: warning: "CONFIG_BT_NIMBLE_MSYS_1_BLOCK_COUNT" redefined
  254 | #define CONFIG_BT_NIMBLE_MSYS_1_BLOCK_COUNT CONFIG_BT_NIMBLE_MSYS1_BLOCK_COUNT

In file included from C:\Users\xxxxx\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.5-b66b5448-v1\esp32s3/include/newlib/platform_include/stdio.h:8,
                 from C:\Users\xxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.3.0\cores\esp32/Arduino.h:27,
C:\Users\xxxxx\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.5-b66b5448-v1\esp32s3/qio_opi/include/sdkconfig.h:812:9: note: this is the location of the previous definition
  812 | #define CONFIG_BT_NIMBLE_MSYS_1_BLOCK_COUNT 12

c:\Users\xxxxx\Documents\Files\Arduino\libraries\NimBLE-Arduino\src/nimconfig.h:289:9: warning: "CONFIG_BT_NIMBLE_TRANSPORT_ACL_FROM_LL_COUNT" redefined
  289 | #define CONFIG_BT_NIMBLE_TRANSPORT_ACL_FROM_LL_COUNT 12

C:\Users\xxxxx\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.5-b66b5448-v1\esp32s3/qio_opi/include/sdkconfig.h:816:9: note: this is the location of the previous definition
  816 | #define CONFIG_BT_NIMBLE_TRANSPORT_ACL_FROM_LL_COUNT 24

```

h2zero, 2025-07-25:
Thanks for reporting, not sure if I will be supporting the new arduino cores though.

h2zero, 2025-07-25:
Yes, I am considering this route, just adds a lot more steps to maintain. In the case for these redefinitions I will likely revert to using the MYNEWT macros instead but that will break everyones configs so will need to be a major release.

PR #1012 (2025-08-10): Cleanup redefinition warnings for arduino core 3.3+
Fixes #1001
closed 2025-08-10 by h2zero