h2zero/NimBLE-Arduino #1008: NimBLEDevice::deinit() causing crash on 2.3.2

justcallmekoko, 2025-08-01:
I encountered an issue with a bit of code I had not had issues with in much earlier versions of NimBLE, such as 1.3.8 for example.
I have created this small sketch to cleanly demonstrate the issue I am experiencing. I am running this on an ESP32-C5-DevKitC-1 v1.2 on Arduino-ESP32 v3.3.0 with NimBLE-Arduino v2.3.2.

The goal with this sketch is to simply initialize NimBLE, prepare a scan callback method that doesn't actually do anything, then deinitialize NimBLE. In the past, this has not been an issue but on 2.3.2, it seems to cause a crash (see Serial Output below).

```C++
#include <Arduino.h>
#include <NimBLEDevice.h>

NimBLEScan* pBLEScan;

// ⚠️ Define the callback object statically
class scanCallbacks : public NimBLEScanCallbacks {
  void onDiscovered(const NimBLEAdvertisedDevice* advertisedDevice) override {
    // No-op
  }
};
scanCallbacks bleCallback;  // <-- static global, never delete

void deinitBLE() {
  Serial.println("Deinitializing BLE...");

  if (pBLEScan != nullptr) {
    if (pBLEScan->isScanning()) {
      Serial.println("Stopping BLE scan...");
      pBLEScan->stop();
      while (pBLEScan->isScanning()) {
        delay(10);
      }
    }

    Serial.println("Clearing results...");
    pBLEScan->clearResults();

    Serial.println("Unbinding callbacks...");
    pBLEScan->setScanCallbacks(nullptr);  // Detach only, DO NOT delete

    delay(10);  // Let BLE stack catch up
  }

  NimBLEDevice::deinit();
  Serial.println("Finished deinitializing BLE");
}

void initBLE() {
  NimBLEDevice::init("demo");
  pBLEScan = NimBLEDevice::getScan();

  pBLEScan->setScanCallbacks(&bleCallback, false);  // Use address of static object
  pBLEScan->setActiveScan(true);
  pBLEScan->setDuplicateFilter(false);
  pBLEScan->setMaxResults(0);
}

void setup() {
  Serial.begin(115200);
  while (!Serial) delay(10);

  initBLE();
  Serial.println("Init BLE complete");

  delay(1000);

  deinitBLE();
  Serial.println("Deinit BLE complete");
}

void loop() {}
```

### Serial Output
```
17:06:09.978 -> Init BLE complete
17:06:10.964 -> Deinitializing BLE...
17:06:10.964 -> Clearing scan results...
17:06:10.964 -> Releasing scan callbacks...
17:06:10.964 -> CORRUPT HEAP: Bad tail at 0x4085c834. Expected 0xbaad5678 got 0xba000000
17:06:10.964 -> 
17:06:10.997 -> assert failed: multi_heap_free multi_heap_poisoning.c:279 (head != NULL)
17:06:10.997 -> Core  0 register dump:
17:06:10.997 -> MEPC    : 0x408007ec  RA      : 0x4080dd28  SP      : 0x4085ed10  GP      : 0x40814c84  
17:06:11.031 -> TP      : 0x4085ef60  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
17:06:11.031 -> S0/FP   : 0x00000001  S1      : 0x4085ee88  A0      : 0x4085ed74  A1      : 0x4081694d  
17:06:11.031 -> A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x4081a000  
17:06:11.064 -> A6      : 0x0000000c  A7      : 0x76757473  S2      : 0x0000007f  S3      : 0x4085ed68  
17:06:11.064 -> S4      : 0x4085ed68  S5      : 0x4085ed74  S6      : 0x00000000  S7      : 0x00000000  
17:06:11.064 -> S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
17:06:11.064 -> T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
17:06:11.064 -> MSTATUS : 0x00001881  MTVEC   : 0x40800003  MCAUSE  : 0x00000002  MTVAL   : 0x00000000  
17:06:11.097 -> MHARTID : 0x00000000  
17:06:11.097 -> 
17:06:11.097 -> Stack memory:
17:06:11.097 -> 4085ed10: 0x00000000 0x00000000 0x4209d770 0x408127b4 0xffffffff 0x00000046 0x0000000a 0x4081694c
17:06:11.097 -> 4085ed30: 0x4085eda0 0x00000018 0x40827858 0x00393732 0xffffffff 0x40816934 0x4209d770 0x40816dd0
17:06:11.097 -> 4085ed50: 0x420952fe 0x40816944 0x4085ed3c 0x40816948 0x420952d8 0x4081694c 0x00000000 0x00000000
17:06:11.097 -> 4085ed70: 0x00000000 0x65737361 0x66207472 0x656c6961 0x6d203a64 0x69746c75 0x6165685f 0x72665f70
17:06:11.131 -> 4085ed90: 0x6d206565 0x69746c75 0x6165685f 0x6f705f70 0x6e6f7369 0x2e676e69 0x37323a63 0x68282039
17:06:11.131 -> 4085edb0: 0x20646165 0x4e203d21 0x294c4c55 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
17:06:11.131 -> 4085edd0: 0x4081a000 0x00000001 0x4085c828 0x4001db2c 0x4085ef60 0x40020d0c 0x42029cca 0x00000200
17:06:11.131 -> 4085edf0: 0x408224e8 0x00000000 0x0000001f 0x4085ee14 0x00000001 0xf7000002 0x00000000 0x40812430
17:06:11.164 -> 4085ee10: 0xfff00000 0x4085c834 0xbaad5678 0xba000000 0x4081a000 0xba000000 0xfff00000 0x00000000
17:06:11.164 -> 4085ee30: 0x00000000 0x00000000 0x00000000 0xbc0abd07 0x40860000 0x00000000 0x00000000 0x00000000
17:06:11.164 -> 4085ee50: 0x4081a000 0x4085c830 0x4085c5a0 0x408121fe 0x4081a000 0x4081a000 0x40821fa8 0x420403a6
17:06:11.164 -> 4085ee70: 0x4081a000 0x4081a000 0x4081a4b0 0x42045184 0x4081a000 0x00000000 0x206d0000 0x42030008
17:06:11.197 -> 4085ee90: 0x4081a000 0x00000000 0x206d0000 0x4204c8f8 0x4081a000 0x00000000 0x40816000 0x42029984
17:06:11.197 -> 4085eeb0: 0x4081a000 0x00000000 0x00000000 0x42013c36 0x4081a000 0x00000000 0x00000000 0x42001644
17:06:11.197 -> 4085eed0: 0x40860000 0x4081ddcc 0x4081a680 0x4002d77c 0x00000000 0x4081ddd4 0x4081a680 0x00000000
17:06:11.231 -> 4085eef0: 0x00000000 0x4081a000 0x40818000 0x4200009e 0x00000000 0x00000000 0x40818000 0x420001fe
17:06:11.231 -> 4085ef10: 0x00000000 0x00000000 0x00000000 0x4201627a 0x00000000 0x00000000 0x00000000 0x4080eb90
17:06:11.231 -> 4085ef30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
17:06:11.231 -> 4085ef50: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
17:06:11.264 -> 4085ef70: 0xbaad5678 0x00000168 0xabba1234 0x0000015c 0x4085edd0 0x00000621 0x40818718 0x40818718
17:06:11.264 -> 4085ef90: 0x4085ef80 0x40818710 0x00000018 0x00000000 0x00000000 0x4085ef80 0x00000000 0x00000001
17:06:11.264 -> 4085efb0: 0x4085cf70 0x706f6f6c 0x6b736154 0x00000000 0x00000000 0x4085ef60 0x00000004 0x00000000
17:06:11.264 -> 4085efd0: 0x00000001 0x00000000 0x00000000 0x00000000 0x00087b79 0x00000000 0x4081b07c 0x4081b0e4
17:06:11.297 -> 4085eff0: 0x4081b14c 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x4201ff38
17:06:11.297 -> 4085f010: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
17:06:11.297 -> 4085f030: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
17:06:11.297 -> 4085f050: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
17:06:11.331 -> 4085f070: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
17:06:11.331 -> 4085f090: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
17:06:11.331 -> 4085f0b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
17:06:11.331 -> 4085f0d0: 0x00000000 0x00000000 0x00000000 0xbaad5678 0x00000068 0xabba1234 0x0000005c 0x00000000
17:06:11.377 -> 4085f0f0: 0x4085f0ec 0x00000000 0x00000000 0x00000000 0x4085f104 0xffffffff 0x4085f104 0x4085f104
17:06:11.377 -> 
17:06:11.377 -> 
17:06:11.377 -> 
17:06:11.377 -> ELF file SHA256: e0cfe4d84
17:06:11.377 -> 
17:06:11.424 -> Rebooting...
```

I have tried keeping it as simple as just using NimBLEDevice::deinit() without the additional NimBLEScan cleanup calls but have experienced the same result.

h2zero, 2025-08-01:
Any chance you can decode that backtrace?

justcallmekoko, 2025-08-01:
I don't have much experience doing that. I have only done it a few times and it was a long time ago.

h2zero, 2025-08-01:
Honestly, your code to deinit isn't required as the call to `NimBLEDevice::deinit();` will do the same thing anyway. Try that and let me know what happens.

h2zero, 2025-08-02:
Okay, I'll look into it, thanks!

h2zero, 2025-08-26:
I have confirmed the issue on an esp32c6 and also confirmed that the cause is an upstream bug as it also happens with the BLE library included in the Arduino core:
```
#include <BLEDevice.h>

void setup() {
    BLEDevice::init("");
    delay(1000);
    BLEDevice::deinit();
}

void loop() {}
``` 

Not sure if I can workaround this or not.

h2zero, 2025-12-03:
@keenravi-hub Unfortunately this is not fixable within this library as it is an upstream bug and will need to wait for the Arduino core to be updated.

justcallmekoko, 2026-01-14:
@h2zero 

Updated to NimBLE 2.3.7 and Arduino ESP32 core 3.3.5 and the following code no longer causes crash. Going to test with NimBLE specifically, then with Marauder source.

```
#include <BLEDevice.h>

void setup() {
    BLEDevice::init("");
    delay(1000);
    BLEDevice::deinit();
}

void loop() {}
```

h2zero, 2026-01-15:
Thanks, I'll test this shortly.