h2zero/NimBLE-Arduino #1011: Typo in parameters of NimBLEScan::start()

LeJoker747, 2025-08-07:
Hi, I just wanted to say that for the parameters of NimBLEScan::start() there is a typo for isContinue variable : the name of the parameter is "[in]" instead of "isContinue" (see picture below). I don't know if it is as typo or not but I would rather write an issue for nothing than just don't say anything.
Have a good day.

[image]

LeJoker747, 2025-08-07:
There is the same problem on some others, like : 
* NimBLEScan::getResults 

[image]

* NimBLEScan::setCallbacks 

[image]

* NimBLEScan::setActiveScan 

[image]

And more.

h2zero, 2025-08-10:
Thanks! I'll address this when I can, also would be happy to accept a PR!

h2zero, 2025-09-02:
Thanks again @LeJoker747, this has been corrected in the latest commits.