h2zero/NimBLE-Arduino #1014: No clean way to change settings on ESP32

sylque, 2025-08-17:
How should I properly change settings on ESP32?
- Changing them in `nimconfig.h` triggers a lot of warnings during compilation (ex: warning: "CONFIG_BT_NIMBLE_MAX_CONNECTIONS" redefined), not mentioning the fact that I need to clone the whole library to change a single file.
- Changing them using the compiler command line or `platformio.ini` doesn't work.

The reason is that, on ESP32, `nimconfig.h` first includes `sdkconfig.h`:
```c
#ifdef ESP_PLATFORM
#include "sdkconfig.h"
#else
```

and `sdkconfig.h` has no conditional macro definition:
```c
// #ifndef CONFIG_BT_NIMBLE_LOG_LEVEL   Missing!
#define CONFIG_BT_NIMBLE_LOG_LEVEL 1
//#endif                                Missing!
```

Hence the issues.

If I were using "ESP-IDF + Arduino as a component", I could use a `sdkconfig.defaults` file to get those constants properly defined in `sdkconfig.h`. But as I'm using plain Arduino, I have no mean to do that.

I'm no expert of the ESP32 toolchain, but I wonder if including `sdkconfig.h` in `nimconfig.h` is a good idea. Why this specific behavior on ESP32?

h2zero, 2025-08-17:
The cause of this is due to the Arduino core now using NimBLE by default instead of bluedroid. The only fix at this time is to use the older Arduino core or Arduino as an IDF component until I refactor this library and go to version 3.x that will break many things.

sylque, 2025-09-07:
Hi @h2zero ,

Maybe I'm doing something wrong, but the issue is still there for me in the latest version, for `MYNEWT_VAL_MSYS_1_BLOCK_COUNT ` and `CONFIG_NIMBLE_MAX_CONNECTIONS`:

```console
Compiling .pio\build\debug\lib710\NimBLE-Arduino\NimBLE2904.cpp.o
In file included from .pio/libdeps/debug/NimBLE-Arduino/src/NimBLE2904.h:21,
                 from .pio/libdeps/debug/NimBLE-Arduino/src/NimBLE2904.cpp:18:
.pio/libdeps/debug/NimBLE-Arduino/src/nimconfig.h:254:9: warning: "MYNEWT_VAL_MSYS_1_BLOCK_COUNT" redefined
  254 | #define MYNEWT_VAL_MSYS_1_BLOCK_COUNT CONFIG_BT_NIMBLE_MSYS1_BLOCK_COUNT
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
<command-line>: note: this is the location of the previous definition
In file included from .pio/libdeps/debug/NimBLE-Arduino/src/nimble/porting/nimble/include/syscfg/syscfg.h:9,
                 from .pio/libdeps/debug/NimBLE-Arduino/src/nimble/nimble/include/nimble/ble.h:25,
                 from .pio/libdeps/debug/NimBLE-Arduino/src/nimble/nimble/include/nimble/hci_common.h:30,
                 from .pio/libdeps/debug/NimBLE-Arduino/src/nimble/nimble/host/include/host/ble_hs.h:30,
                 from .pio/libdeps/debug/NimBLE-Arduino/src/NimBLELocalValueAttribute.h:27,
                 from .pio/libdeps/debug/NimBLE-Arduino/src/NimBLEDescriptor.h:24,
                 from .pio/libdeps/debug/NimBLE-Arduino/src/NimBLE2904.h:24:
.pio/libdeps/debug/NimBLE-Arduino/src/nimble/esp_port/port/include/esp_nimble_cfg.h:183:9: warning: "MYNEWT_VAL_BLE_MAX_CONNECTIONS" redefined
  183 | #define MYNEWT_VAL_BLE_MAX_CONNECTIONS (CONFIG_NIMBLE_MAX_CONNECTIONS)
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
<command-line>: note: this is the location of the previous definition
```

My `platformio.ini` file:
```
build_flags = 
  ...
  -D MYNEWT_VAL_BLE_MAX_CONNECTIONS = 5
  -D MYNEWT_VAL_MSYS_1_BLOCK_COUNT = 48
  -D MYNEWT_VAL_BLE_HS_LOG_LVL = 2
  -D MYNEWT_VAL_NIMBLE_CPP_LOG_LEVEL = 2
  -D MYNEWT_VAL_NIMBLE_CPP_DEBUG_ASSERT_ENABLED = 1
  -D MYNEWT_VAL_NIMBLE_CPP_ENABLE_RETURN_CODE_TEXT
  -D MYNEWT_VAL_NIMBLE_CPP_ENABLE_GAP_EVENT_CODE_TEXT
  -D MYNEWT_VAL_NIMBLE_CPP_ENABLE_ADVERTISEMENT_TYPE_TEXT
```

h2zero, 2025-09-07:
You don't seem to be on the latest commit:
`pio/libdeps/debug/NimBLE-Arduino/src/nimble/esp_port/port/include/esp_nimble_cfg.h:183`
this file has been removed, please delete your pio folder.

sylque, 2026-03-14:
> You don't seem to be on the latest commit:
>pio/libdeps/debug/NimBLE-Arduino/src/nimble/esp_port/port/include/esp_nimble_cfg.h:183
>this file has been removed, please delete your pio folder.

@h2zero, I believe I’ve found why the issue described [above](https://github.com/h2zero/NimBLE-Arduino/issues/1014#issuecomment-3264025702) keeps coming back:

While the file `esp_nimble_cfg.h` has indeed been removed from the main branch, it appears it is still physically present in **release tags**. See for example in v2.3.9:
https://github.com/h2zero/NimBLE-Arduino/blob/2.3.9/src/nimble/esp_port/port/include/esp_nimble_cfg.h