h2zero/NimBLE-Arduino #1019: reducing power consumption on an esp32-c3

icodk, 2025-08-26:
This is more of an advice request then an issue. Hope you can help me on that.
I developed my application on a dev board with ESP32 WROOM without any issue. I then recompiled it to 
an esp32-c3 and it runs as expected. However I noticed that the board/chip get relatively hot in comparison
with the WROOM board. I use only BLE , I2C and a single GPIO PIN for reading a DS18b20 temp. sensor but no Wifi. 
The BLE continuously  advertises when not connected, which is most of the time.
It also constantly do (active) scanning .
Only once a minute, reads CTS info from the server if connected.
One of the advices that I got from AI to reduce power consumption is to:
1. Increase connection interval & slave latency.
2. Lower TX power if your stack/API allows.
Could you explain how can I do it with the NimBLE library which I am using extensively(with a great pleasure) ?
Do you have any other ideas for reduce power consumption in general or specific for use with your library
I am using Arduino IDE 2 and the NimBLE library 

Thanks

h2zero, 2025-08-26:
If your application is constantly running scan/advertising then there isn't much you can do that will have any great impact. You could reduce the CPU frequency to 80Mhz and increase the advertising interval on top of the suggestions from AI, that's about all I have for this case.

h2zero, 2026-04-10:
Sorry I didn't get back to you earlier @icodk, I hope you found the function.
For future reference it is here https://h2zero.github.io/NimBLE-Arduino/class_nim_b_l_e_server.html#a56add4759f7df045f51b70a026897975

closed 2026-04-10 by h2zero