h2zero/NimBLE-Arduino #1030: Compatibility with A2DP?

Nawor3565, 2025-09-09:
Hi there! I'm working on a project that uses both BLE to send/receive commands and Bluetooth "classic" for streaming audio over A2DP (currently implemented with Phil Schatzmann's [ESP32-A2DP](https://github.com/pschatzmann/ESP32-A2DP) library). I had success using the default ESP32 Arduino BLE stack alongside A2DP, but it seems like NimBLE is [doing something different](https://github.com/pschatzmann/arduino-audio-tools/discussions/1228#discussioncomment-8237689) that makes it impossible to have both A2DP and BLE initialized at once. 

I'm going to look into this more in the meantime because the performance gains from NimBLE are worth the effort, but by any chance is there an easy fix that I'm overlooking?

Thanks!

h2zero, 2025-09-09:
A2DP is classic bluetooth, which is not supported by the NimBLE stack. 
https://github.com/pschatzmann/ESP32-A2DP/discussions/672

Nawor3565, 2025-09-09:
Crap. Well, thanks for the heads up!

closed 2025-09-09 by Nawor3565