h2zero/NimBLE-Arduino #1042: Impossible to getService, always null pointer

JPFrancoia, 2025-09-29:
Hello,

I have been trying to connect to a GATT server from my XIAO ESP32S3plus chip.

Here is the code I'm using:

```cpp
#include "NimBLEDevice.h"

void setup() {
    Serial.begin(115200);

    NimBLEDevice::init("ardoise_esp32");

    Serial.println("Starting BLE scan...");
}

void scanAndConnect() {
    NimBLEScan *pScan = NimBLEDevice::getScan();
    NimBLEScanResults results = pScan->getResults(10 * 1000);

    Serial.print("Devices found: ");
    Serial.println(results.getCount());

    // NimBLEUUID serviceUuid("12341000-1234-1234-1234-123456789abc");
    NimBLEUUID serviceUuid((uint16_t)0x1805);

    for (int i = 0; i < results.getCount(); i++) {
        const NimBLEAdvertisedDevice *device = results.getDevice(i);

        if (device->isAdvertisingService(serviceUuid)) {
            if (!device->isConnectable()) {
                Serial.println("Device not connectable, skipping...");
                continue;
            }
            Serial.print("Found target service on device: ");
            Serial.println(device->toString().c_str());
            Serial.println("Connecting...");

            NimBLEClient *pClient = NimBLEDevice::createClient();
            pClient->setConnectTimeout(5000);
            // pClient->setConnectionParams(12, 12, 0, 51);  // Example: min/max
            // interval 15ms, latency 0, timeout 510ms

            if (!pClient) {  // Make sure the client was created
                Serial.println("Failed to create client");
                break;
            }

            int connectResult = pClient->connect(device, true, true, false);
            Serial.printf("Connect result: %d\n", connectResult);

            if (connectResult) {
                Serial.println("Connected to device");

                if (device->haveServiceUUID()) {
                    NimBLEUUID devUUID = device->getServiceUUID();
                    Serial.print("Found ServiceUUID: ");
                    Serial.println(devUUID.toString().c_str());
                    Serial.println("");

                    // String serviceData = String(device->getServiceData(serviceUuid).c_str());
                    // Serial.print("Service Data: ");
                    // Serial.println(serviceData);
                    bool legacy = device->isLegacyAdvertisement();
                    Serial.print("Legacy: ");
                    Serial.println(legacy ? "true" : "false");

                }

                pClient->discoverAttributes();
                if (!pClient->discoverAttributes()) {
                    Serial.println("Service discovery failed");
                    NimBLEDevice::deleteClient(pClient);
                    return;
                }

                // // // Force discovery of services
                // std::vector<NimBLERemoteService *> services =
                //     pClient->getServices(true);
                // if (services.empty()) {
                //     Serial.println(
                //         "Failed to discover services or no services found");
                //     NimBLEDevice::deleteClient(pClient);
                //     return;  // or continue
                // }
                // Serial.printf("Services discovered: %d\n", services.size());

                NimBLERemoteService *pService =
                    pClient->getService(serviceUuid);

                Serial.println("Getting service...");
                Serial.printf("Service: %p\n", pService);

                if (pService != nullptr) {
                    NimBLERemoteCharacteristic *pCharacteristic =
                        // pService->getCharacteristic("2A6E");
                        pService->getCharacteristic("2A2B");

                    if (pCharacteristic != nullptr) {
                        std::string value = pCharacteristic->readValue();
                        Serial.print("Characteristic value: ");
                        Serial.println(value.c_str());
                        // print or do whatever you need with the value
                    }
                } else {
                    Serial.println("Failed to find service");
                }
            } else {
                // failed to connect
                Serial.println("Failed to connect to device");
                Serial.printf("Connection failed, error: %d\n",
                              pClient->getLastError());
            }

            NimBLEDevice::deleteClient(pClient);
        }
    }
}

void loop() {
    delay(10000);  // Wait 10 seconds before the next scan
    scanAndConnect();

    Serial.println("Scan complete. Waiting 10 seconds before next scan...");
}
```

When I run this code on the board, I'm gettin:

```
Devices found: 42
Found target service on device: Name: , Address: 45:19:c6:ed:e9:d2, serviceUUID: 0x1805
Connecting...
Connect result: 1
                 Connected to device
Found ServiceUUID: 0x1805

Legacy: true
Service discovery failed
```

When I'm not running `discoverAttributes` and go straight to this line:

```
                NimBLERemoteService *pService =
                    pClient->getService(serviceUuid);
```

I always end up with a null pointer.

I tried running two different servers:
- one from my computer, with python-bluezero. I can connect to it and get the value of one characteristic with nrf connect
- one from my phone, through nrf connect

I end up with the same result in both cases.

I even tried checking 16 bits vs 128 bits uuid, with no success.

Would you have any idea about what's happening?

h2zero, 2025-09-29:
The server might not support getting the individual service so you will need to discover all attributes.

h2zero, 2025-09-29:
Can you show me the debug level log output?

JPFrancoia, 2025-09-30:
Hey, thanks for your answer. Here you are:

[screenlog.txt](https://github.com/user-attachments/files/22613850/screenlog.txt)

There are two scans in these logs. The logs from Nimble are mixed with the logs from my own code, but maybe this is the interesting bit:

```
I NimBLEScan: New advertiser: 70:a8:d3:7e:01:75
D NimBLEScanCallbacks: Discovered: Name: , Address: 70:a8:d3:7e:01:75, appearance: 1344, serviceUUID: 12341000-1234-1234-1234-123456789abc
D NimBLEScanCallbacks: Result: Name: , Address: 70:a8:d3:7e:01:75, appearance: 1344, serviceUUID: 12341000-1234-1234-1234-123456789abc
[...]
I NimBLEScan: New advertiser: f5:62:e1:e4:1b:4e
D NimBLEScanCallbacks: Discovered: Name: , Address: f5:62:e1:e4:1b:4e, manufacturer data: 4c00121960a71fbef9d0fba1c8cb24059f5b8262fe1c7f1dc70fa6004e
D NimBLEScanCallbacks: Result: Name: , Address: f5:62:e1:e4:1b:4e, manufacturer data: 4c00121960a71fbef9d0fba1c8cb24059f5b8262fe1c7f1dc70fa6004e
I NimBLEScan: New advertiser: 4b:50:de:5b:93:9c
D NimBLEScanCallbacks: Discovered: Name: , Address: 4b:50:de:5b:93:9c, manufacturer data: 4c001006301e25b78c42, txPower: 12
D NimBLEScanCallbacks: Result: Name: , Address: 4b:50:de:5b:93:9c, manufacturer data: 4c001006301e25b78c42, txPower: 12
D NimBLEScan: discovery complete; reason=0
D NimBLEScanCallbacks: Scan ended; reason 0, num results: 35
Devices found: 35
Found target service on device: Name: , Address: 70:a8:d3:7e:01:75, appearance: 1344, serviceUUID: 12341000-1234-1234-1234-123456789abc
Connecting...
D NimBLEClient: >> connect(70:a8:d3:7e:01:75)
Connect result: 1
Connected to device
Found ServiceUUID: 12341000-1234-1234-1234-123456789abc

Legacy: true
E NimBLEClient: Disconnected, could not retrieve services -aborting
Service discovery failed
D NimBLEClient: >> getService: uuid: 12341000-1234-1234-1234-123456789abc
E NimBLEClient: Disconnected, could not retrieve services -aborting
D NimBLEClient: << getService: not found
Getting service...
Service: 0x0
Failed to find service
Scan complete. Waiting 10 seconds before next scan...
```

h2zero, 2025-09-30:
Thanks, what is the server device? Obviously the issue is you're getting disconnected, not sure why though. Do you have another esp32 to test this with, sometimes this happens due to bad hardware.

h2zero, 2025-09-30:
That is odd indeed, what happens if you just establish a connection and loop, how long does it take to disconnect?

JPFrancoia, 2025-09-30:
With which configuration should I try this?

h2zero, 2025-09-30:
Cool, now check with the nimble code and please enable debug logging, hopefully we can see something.

JPFrancoia, 2025-09-30:
I think I got it now.

I had:

```cpp
int connectResult = pClient->connect(device, true, true, false);
```

And according to the docs, that means that the connection will be made asynchronously and that the call won't block. My guess is that `connectResult` is true in this case, but the client might not always be really connected?

Using:

```cpp
            int connectResult = pClient->connect(device, true, false, false);
```

Gets me a reliable connection all the time now.

h2zero, 2025-09-30:
Oh wow, I didn't even see that! I should have caught that as well, sorry about that. Glad it's sorted :) can we close this now?